FlightOne/FalcoX-Issue-Tracker #283: Target NBDINFINITYF4 Bug

PanicoFPV, 2021-11-16:
The Target NBDINFINITYF4 has a bug that will not allow user to get through the radio wizard. It is not picking up Crossfire (FW 6.09 or 6.10) . Manually updating UART to RX does not work either. 

**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

WildWillyFL1, 2021-11-17:
What UARTs have you tried hooking it up to?

PanicoFPV, 2021-11-17:
Hi- 

It was UART 3. 

Richard Panico

WildWillyFL1, 2021-11-18:
Are there any other non-inverted UARTs on that FC you can try hooking it up to test?

PanicoFPV, 2021-11-18:
Hi Willy- 

I’m not sure to be honest. Looks like UART 1 & 6 are taken by DJI? 

Richard Panico

PanicoFPV, 2021-11-18:
I don’t know how to tell whether it’s uninverted other than the diagram I just sent. If I need to look for something on the FC that will determine that, just lemme know. 

Richard Panico

PanicoFPV, 2021-11-19:
Just following up here. Anything I can do to test something out? Please lemme know. 

Richard Panico

PanicoFPV, 2021-11-23:
Good day- 

Just following up again. Can someone please help a brotha out? 

Richard Panico

WildWillyFL1, 2021-11-24:
> Good day- Just following up again. Can someone please help a brotha out? Richard Panico
> […](#)

Yeah can you try hooking just the crsf tx line to any uninverted tx or rx pad on the FC and try those, it should work on any non-inverted UART

PanicoFPV, 2021-12-02:
Me again-

I still cannot flash my NBD Infinity 200 FC.  I'm not sure what the issue
is. I'm not seeing where there is an uninverted pad according to NBD's
website (see pic below). Are you saying i need to rewire the RX to get it
to work because in BF & EMU my rx is listed and wired to UART 4. I believe
I mistakenly told you UART 3 in a previous email, my apologies for that.
However, I had been trying UART 4 and that didn't work. Can you please let
me know how I should proceed?

[image: image.png]

closed 2022-01-08 by stale[bot]