FlightOne/FalcoX-Issue-Tracker #285: Setup Wizard fails to recognize CRSF on Racepit

oscarkim007, 2021-11-18:
**Describe the bug**
Flashed a Racepit F4 with FalcoX Alpha latest firm but Setup fails to recognize the CRSF (soldered to the designated tx1,rx1 pad). 

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

PGIII, 2021-11-18:
Duplicate of #282

closed 2021-11-18 by PGIII

Fianco182, 2021-12-14:
Haii, i have a the same issues, i had a Racepit F4 and try to flash firmware with the falcoX , but now the csfr not detected , i use betafpv elrs 900, in betaflight / emuflight can detected or used, but in falcox cannot detected the csfr, i followed the wiring Tx to RX FC , RX to TX fc, but still not working, but i havent try the inverted flash. 
can u help me guys ? 🥲