FlightOne/FalcoX-Issue-Tracker #287: Target FOXEER F722 V2 Issue detecting VTX

VitalFPV, 2021-12-03:
I tryed  FalcoX 10.1.1.5818, and also tryed 10.1.1.5792, and even 10.1.0.5490. 

i use Foxeer F722 v2 mini FC, and my target is Foxeer F722 V2

My bug is that Falco Wizzard does not detect VTX, i use TBS Unify Nano 32 PRO 500mW with Smart Audio 2.1. 
even if i passthrough wizzard, i manually set up SmartAudio on Uart5. then return to VTX tab, still nothing, tryed to pick manually every single option that FalcoX has, and no success. Smart Audio 2.1 dont want to work...  had some aliexpress cheap VTX with smart audio support and everything worked fine, same RIG, nothing has changed, just desoldered pin from old vtx s.a. port to the new VTX TBS Unify nano 32 Pro and BAAAAAAAM! Sh#t happens....

d00m303, 2021-12-07:
did u tried switching pro32nano to SA from CRSF? i think they come set to CRSF from default. Im using this vtx and all works fine with falco thru SA ;)

VitalFPV, 2021-12-11:
> did u tried switching pro32nano to SA from CRSF? i think they come set to CRSF from default. Im using this vtx and all works fine with falco thru SA ;)

Hey d00m303, yes today i tryed locking/unlocking VTX, sitching manually pushing button between OFF/CRSF/SA/PWM** (like in the TBS table)  and nothing helped me. i guess i will just solder my SA wire o crossfire wire, and setup my VTX from my RADIO :) i think its a bug of target or board. becaus no S.A. worked on betaflight, or even emuflight...

closed 2022-01-11 by VitalFPV