FlightOne/FalcoX-Issue-Tracker #289: Profile with BUG arming the drone. H7 10.1.1.5829

Leoparna, 2021-12-09:
h7 controller latest fw version.
1- I created 2 rate profiles (very different) with identical pid and filtering, tpa....
2- key for profile 1 (low rate) and 2 (fast rate)
3- I also put osd information other than profile 1 (sat, lat and long) and 2 (no gps information).
4- key selected for profile 1, gps information on screen and when arming the drone, gps information disappears. The drone is also skittish as if I were switched to profile 2... and I'm not.
5- When arming the drone switched to profile 1, it changes itself to profile 2. Final test on the flightone app screen, when arming the drone it changes the profile by itself without touching the switch
6- The same problem happens with other friends who also tested different profiles

PGIII, 2021-12-09:
Do you have "Profile For:" set to all cells on all of the profiles ?

Leoparna, 2021-12-10:
> Do you have "Profile For:" set to all cells on all of the profiles ?

Use 6s. The 2 profiles have 6 cells

PGIII, 2021-12-10:
Set them to all cells that is for making the profiles auto switch based on the battery that’s plugged in

Leoparna, 2021-12-10:
> Set them to all cells that is for making the profiles auto switch based on the battery that’s plugged in

OK. I will take the test. thanks

Leoparna, 2021-12-14:
> Set them to all cells that is for making the profiles auto switch based on the battery that’s plugged in

it worked very well....thanks for the help

closed 2021-12-14 by Leoparna