FlightOne/FalcoX-Issue-Tracker #297: Motor Spins Abnormally before Quad connects with Transmitter

Jurisjury, 2022-01-21:
This Quad works fine until this issue happened. 
Happens only for the first 2-4 Battery Plug ins.
And during Plug-in if issue doesn’t occur, quad works great.

Quad Electronic  Details: 
T-Motor F7 HD And PRO II Stack
DJI Caddx Air Unit
T-Motor 2307 F40 PROII

FalcoX:
Version 10.1.1.5818
using DSht600

Tried flashing it with Betaflight to see if there is ESC/FC Problem and problem doesn’t occur.

Flight One, please have a look at this. This is a major safety Concern. I was lucky I didn’t cut my fingers during the first time it happened.

Flight Video of Quad before Problem: https://youtu.be/_1QZFY35e2s

PGIII, 2022-01-24:
Is the ESC Blheli_s or Blheli_32 and what version is on them ? also can you replicate the issue with DJI MSP disabled in the FC

Jurisjury, 2022-02-08:
It is BLheli_32 32.7

closed 2022-03-17 by stale[bot]