FlightOne/FalcoX-Issue-Tracker #299: DJI osd  Voltage

mickeyboss123, 2022-02-17:
**Version**
FLX_IFLIGHTF722_TWING_10.1.1.5850

**Target**
DJI osd  Voltage

**Describe the bug**
I correctly set uarts to msp (dji osd), Arm Status is working properly, Voltage And Cell Volt is showing 0.0v
[image]
[image]
[image]

PGIII, 2022-03-17:
Is the voltage displayed correctly in the configurator ?

mickeyboss123, 2022-04-20:
The same shows 0.0v

PGIII, 2022-04-23:
Probably and issue with the target then ill have to check the settings for that target

mickeyboss123, 2022-04-25:
Thank you for your help