FlightOne/FalcoX-Issue-Tracker #304: iFlight SucceX F7 TwinG BT still not working

jokker87, 2022-04-03:
**Version**
10.1.1.5925

**Target**
IFlight F722 TwinG

**Describe the bug**
After 10.1.1.5850 all motors are spinning in the wizard, but in "normal mode", only 3 of 4 motors are spinning. The mapping in BF is as follows:

C06: MOTOR 2
C07: MOTOR 4
C08: MOTOR 1
C09: MOTOR 3

and MOTOR 1 is not spinning in FalcoX.

**To Reproduce**
Flash newest Firmware on Target and finish wizard and then arm. Only 3 of 4 motors are spinning.

**Expected behavior**
All 4 motors should spin like in the wizard.

jokker87, 2022-04-03:
Same problem as in #116 and #169

jokker87, 2022-05-03:
bump, it's still not fixed

PGIII, 2022-05-03:
Can you get a dump from the board after its setup

closed 2022-06-09 by stale[bot]