FlightOne/FalcoX-Issue-Tracker #312: FC does not boot after flashing

TcheTcheFPV, 2022-08-09:
**Version**
10.2.6.6102

**Target**
Xrotorf7conv

**Describe the bug**
I was running a Xrotorf7conv FC without any problem until it fries. So I have replaced the FC with another Xrotorf7conv for which I get a new license, and since I am unable to get it working. I can flash in DFU, then flash the firmware, but once I plug the battery, I get only 3 beeps instead of 5, and the CRSF is not detected, so the wizard is not finishing. 
It is working with betaflight.

**To Reproduce**
Flash the FC with the last configurator alpha and last alpha firmware

PGIII, 2022-08-09:
What Gyro does the board have on it ?

TcheTcheFPV, 2022-08-09:
According to the manufacturer, it's a ICM20602.

PGIII, 2022-08-11:
can you get a picture of both sides of the board, also when you flash it is it showing the correct target when it connects to the configurator ?

TcheTcheFPV, 2022-08-11:
here is the screenshot when pluging the FC in DFU, and then when flashing is finished:

[image]

[image]

And here are the pictures of the board

[image]

[image]

TcheTcheFPV, 2022-08-12:
Hi @PGIII ,

I have discovered something. Using emulflight to get the gyro type, it seems that my new board has another gyro.

Old FC which is like the manufacturer specs:
`# version
# EmuFlight / HOWI_HOBBYWING_XROTORF7CONV (S7X2) 0.4.1 Jul 29 2022 / 13:28:00 (7bac6ddd0) MSP API: 1.51

# status
System Uptime: 14 seconds
Current Time: 2022-08-12T11:47:20.831+00:00
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=216MHz, Vref=3.32V, Core temp=42degC, GYRO=ICM20602, ACC=ICM20602
Stack size: 2048, Stack address: 0x20010000
I2C Errors: 5, config size: 2794, max available config: 16384
CPU:4%, cycle time: 127, GYRO rate: 7874, RX rate: 33, System rate: 9
Arming disable flags: RXLOSS CLI MSP`

And the NEW FC which has different gyro:
`# version
# EmuFlight / HOWI_HOBBYWING_XROTORF7CONV (S7X2) 0.4.1 Jul 29 2022 / 13:28:00 (7bac6ddd0) MSP API: 1.51

# status
System Uptime: 16 seconds
Current Time: 2022-08-12T11:48:01.060+00:00
Voltage: 0 * 0.1V (0S battery - NOT PRESENT)
CPU Clock=216MHz, Vref=3.30V, Core temp=38degC, GYRO=MPU6500, ACC=MPU6500
Stack size: 2048, Stack address: 0x20010000
I2C Errors: 5, config size: 2794, max available config: 16384
CPU:4%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Arming disable flags: RXLOSS CLI MSP`

So they are now using a MPU6500 instead of a ICM20602.

Is it possible to get FalcoX working with the MPU6500 on this board ?

PGIII, 2022-08-17:
[FLX_XROTORF7CONV_10.2.7.6205.flx.zip](https://github.com/FlightOne/FalcoX-Issue-Tracker/files/9364354/FLX_XROTORF7CONV_10.2.7.6205.flx.zip)
You can try this one I enabled the 6500 on it, haven't heard of a board having that gyro on it in a while

TcheTcheFPV, 2022-08-17:
Hi @PGIII ,

Thank you very much, it is working ! I could flash it and configure it perfectly. Now I have to test in flight.

It seems that some manufacturer and modifying some components without notice due to lack of components. I hope this MPU6500 is not too bad...

TcheTcheFPV, 2022-08-24:
Hi @PGIII ,

It flies as expected : perfectly

I hope you can integrate this gyro in this target for now on.

Thank again for your help :)

closed 2022-09-02 by TcheTcheFPV