FlightOne/FalcoX-Issue-Tracker #313: Buzzer on MillivoltOSD is constantly beeping

epcim, 2022-08-09:
I am using one similar to https://www.banggood.com/cs/HGLRC-Soter-22x12x12mm-Super-Loud-Buzzer-5V-with-LED-For-FPV-Racing-RC-Drone-p-1831253.html?cur_warehouse=CN&rmmds=search

3 wires

issue: start beeping after battery is plugged in

is there any option to set on terminal in order give it signal (invert behaviour) - (so not beeping immediately battery is plug in?)

epcim, 2022-08-11:
Later find out, that it beeps wrong (on battery plug-in event) but when I later Arm and fly. When it lands the buzz works as expected (not beeping constantly, will start beep when battery is unplugged, and beeps on signal sent through RF)

Seems like, the bug happens, only when FC is started the pin is not in correct state. (not measured yet)

For others docs, to wire 3wire buzzer is here: https://support.flightone.com/index.php/knowledge-base/connecting-a-buzzer/

PGIII, 2022-08-11:
How do you have the buzzer attached ? are you using pin toggle and the buzzer is on TX4 or do you have it powered from the buzzer pads ?

closed 2022-09-18 by stale[bot]