FlightOne/FalcoX-Issue-Tracker #318: [NEW TARGET] New Target For Firmware MAMBA MK4 F722 APP

ujenmr, 2023-02-09:
Hi, 
Is it possible to add firmware compatibility for the FC MAMBA MK4 F722 APP?

ujenmr, 2023-02-09:
Hi, @WildWilly37 can you look?

PGIII, 2023-02-09:
https://www.diatone.us/products/mb-mk4-f722-app-fc Is this the FC ?

ujenmr, 2023-02-09:
Yes, BF

# Betaflight / STM32F7X2 (S7X2) 4.4.0 Feb  1 2023 / 23:57:34 (4605309d8) MSP API: 1.45
board_name MAMBAF722_2022B
manufacturer_id DIAT

PGIII, 2023-02-09:
Looks like it uses gyros we don't have support for right now. The ICM one I have a test branch for so can probably make a target that supports that one.

ujenmr, 2023-02-09:
oo cool, if you give me I can test and report the result

ujenmr, 2023-02-09:
very strange, but on the target MambaF7 FC didn't load, for example, it didn't find RX in the configurator, when I connected the battery, even BLHeli didn't send a communication signal with FC

PGIII, 2023-02-10:
It has a gyro that's not supported yet, if the gyros not working nothing is going to work correctly. Ill let you know when I have new FW you can try

pcviruspcvirus, 2023-02-17:
Блин-Блин, мне тоже нужна прошивка. Купил ключ, теперь сижу и плачу)))

pcviruspcvirus, 2023-02-17:
I need firmware too. I bought a key, and as it turned out later, the firmware does not support this flight.

ujenmr, 2023-02-17:
I'm a fan of falcox, can't find anything like this, BF or kiss not the same taste... @PGIII Release F1 FC i will be first in line to buy